bluebandit21/magic-wifi #29: Need more information

D4rk4, 2023-08-25:
Could you please enhance the readme by providing details on the range, hardware specifications, documentation, and actual speed? It would be helpful if you could also mention the specific hardware components used and include images of the actual hardware. Do you have an introductory video about your ethernet bridge on YouTube?

bluebandit21, 2023-08-25:
Hi Dmitry!
Good to hear from you :)

We can definitely fill out the documentation more and add some images over the next few days.
We unfortunately don't have any videos right now actually describing the project, although we have a few phone recordings showing our prototype working and such 🙂 

If you don't mind me asking, are you interested in potentially recreating this yourself?
If so, our team might be able to give you some information that might be helpful :)

D4rk4, 2023-08-26:
Thank you for the update. I would like to try assembling your project, but I couldn't find instructions on how to compile the firmware, for instance.

bluebandit21, 2023-08-27:
Ah, very good point :)

So, this project was developed targeting an MSP430FR2355 board, which is Texas Instruments's.
To build the project, we used TI's CCStudio -- https://www.ti.com/tool/CCSTUDIO

All the random directories in the repo with a ".cssproject" file are CCStudio projects -- in particular, the one under "magicwifi" is the one you probably actually want, as it's the final project and not a random development prototyping thing.

You'll likely want to buy a MSP430FR2355 launchpad for development (although there's a bit of a funny thing in that a random pin you'll need isn't actually exposed to something usable, so you'll have to solder something onto the launchpad); from there it's really easy to build, debug, and load the code -- you just plug it into your computer with the provided USB cable and CCStudio does the rest of the work for you.

If you want to actually build the project on a PCB, you'll need a TI JTAG flashing thingy. I don't remember the details off the very top of my head right now, but can look them up for you if you're interested.

I'll add ^^ in the README.md as soon as I can, but hopefully this might be enough to get started :)

bluebandit21, 2023-08-27:
I will say, fwiw, that the project isn't really in a production-ready state -- 
We found that there was some problem in the LoRa antenna designs of the PCB that led to them having significantly less gain than they should have while testing.
However, we didn't get a chance to go back and debug the design to find out what actually went wrong.

Building a few prototype systems with the launchpads I mentioned above and dev component modules may not be a bad idea before trying to use our PCB :)

I can't remember if we included a BOM in this repo, if not I can copy over all non-sensitive info from our final one so you know what you'd need to order from DigiKey for either approach.

bluebandit21, 2023-09-08:
[TEMP - Sheet1.pdf](https://github.com/bluebandit21/magic-wifi/files/12554278/TEMP.-.Sheet1.pdf)

^ is the list of components we ended up purchasing, including both for our dev boards and the production PCBs :)
The links in the PDF for the left section work and will take you to the places you can buy them.

Is there anything else you need? :)

bluebandit21, 2023-09-15:
@D4rk4 
Was that enough to get you going? :)