bluebandit21/magic-wifi #33: Redesign LoRa Antenna

Santa-Bunny, 2024-01-10:
The antenna routing for the LoRa antennas struggles to actively produce enough power with minimal feedback. This is reducing our max LoRa distance to ~100m instead of the theoretical km we could be getting with better routing. A new design must be decided upon for the next version of the PCB if this product is to be viable in any sense. As of right now we are almost only using WiFi because of the LoRa fault.

bluebandit21, 2024-01-12:
RE: Bad antenna design --
I've got a physical board sitting in a drawer if taking any physical measurements with an oscilloscope or the like could be helpful :)

Thoughts?

Santa-Bunny, 2024-01-14:
I might need that, at the moment I'm focused on getting the rest of the PCB laid out again before I move my attention to making anything new.